stanleyhuangyc/Freematics #7: logger & elm327 mode not working

hayden-t, 2015-04-03:
HI again. With the elm327 sketch i can connect and get a response from 0100 via bt terminal, but using app "E OBD2 Facile -Car Diagnostics" it can connect but not read any data. But with non-logger elm327 off ebay it works fine.

Im trying to debug to work out whats going on, as it doesnt work in logger mode either. So i thought if i could get it working nicely in elm327 mode that would be a good start.

In the elm327 sketch can you explain what the purpose of this code bit is ?
```
        if (cmdlen == 3 && !memcmp(cmd, "ATZ", 3)) {
          delay(1000);
          while (Serial.available()) Serial.read();
          Serial.println("ELM327 v1.4b");
        }
```

Are there any other tests / ideas you might have to work out what's up?

hayden-t, 2015-04-04:
ok, so after enabling debugging in the obd library, this is the output: 
```
ATZ
OBDUART v1.0
ATE0
ATE0
OK
ATL1
OK
ATSP5
OK
010C
BUS
INIT:
...ERROR
```
(I get same output, if auto protocol selected too)
I will now do the same init test with it the logger in elm327 mode, and with a standalone elm327

hayden-t, 2015-04-04:
here are results from standalone elm327:
```
ATZ
ELM327 v1.5
ATE0
OK
ATL1
OK
ATSP5
OK
010C
BUS INIT: OK
41 0C 00 00
7F 01 12
```
i will now try data logger with elm mode sketch

hayden-t, 2015-04-04:
results from elm327 sketch on logger: (i sent 010C a few times to show varying response)
```
ATZ
ELM327v1.4b
SEARCHING
7F F5 12
ATE0
ATE0
OK
ATL1
OK
ATSP5
OK
010C
BUS
INIT:
...ERROR
010C
BUS
INIT:
...OK
41 0C 00 00
010C
7E
010C
7E 94 84 F1 10 41 0C 00 00
```
so hmm, definitely something going on ....

hayden-t, 2015-04-05:
I did another test to rule out any possible problems with SoftwareSerial, by using the elm327 sketch with sd logging instead of bt serial output. and this is the output, once again, broken variable output, i really think there is something wrong with your OBD interface chip, not the arduino side of things.
```
STARTUP
010C
BUS
INIT:
...OK
41 0C 00 00 

>010C
E2 03 61 8D 

>010C
7E 

>010C
E2 03 61 8E 

>010C
7F 01 12 

>010C
E2 03 61 8F 

>010C
7E 94 84 F1 10 41 0C 00 00 

>010C
7E 94 00 00 07 E2 03 61 8F 02 84 F1 10 41 0C 00 00 

>
```
To get this, at startup, i issue standard init commands: ATZ, ATE0, ATL1, ATSP5, then request 010C every 2 seconds.

stanleyhuangyc, 2015-04-05:
SoftwareSerial is not stable and loses data on baud rate higher than 9600bps.

Regards,
Stanley

hayden-t, 2015-04-05:
Great to hear from you ! It sounds plausible, but why then do you supply it with bt 2.1 which runs at 38400 ?? is there a way to reduce this to 9600 ?

stanleyhuangyc, 2015-04-05:
You can add slight delay() after sending every character. Actually 38400 is still OK for not so intensive data transmission.

Regards,
Stanley

hayden-t, 2015-04-05:
actually i remember now changing a bt modules baud before: http://blog.oscarliang.net/multiwii-bluetooth-change-baud-rate/

I did see your use of delays in the elm sketch and was starting to wondered if thats what it was about.

mind you, as i have pointed out above, even not using softwareserial and logging to sd card instead still did not work !?

would you regard your ENABLE_DATA_OUT option as being too data intensive then ?

hayden-t, 2015-04-05:
Still interested in your response to my last  post, but can you tell us what the "BT Module Key" pin is on your BT 2.1 module ? (and maybe link to a datasheet)

hayden-t, 2015-04-05:
Ok, Im stuck, unless you can get this basic example working, I will be seeking a refund (or exchange if you think my unit is faulty). Its disappointing, but I feel nothing else left to try.

Here is my complete test sketch, NOT using SoftwareSerial to rule that out, simply logging the results to SD. The 2 posts after the sketch show the output from the sketch then the output from the same commands but with a standard elm327 adapter off ebay.

hayden-t, 2015-04-05:
```
#include <SPI.h>
#include <SD.h>

File myFile;

void setup()
{
  
  pinMode(SS, OUTPUT);
  SD.begin(10);
  myFile = SD.open("test.txt", FILE_WRITE);
  myFile.print("Starting Test\r");
  myFile.flush();
  
   Serial.begin(38400);

   delay(1000);
   Serial.write("ATZ\r");
   delay(2000);
   while(Serial.available())myFile.write(Serial.read());
   myFile.flush();

   delay(1000);
   Serial.write("ATE1\r");
   delay(2000);
   while(Serial.available())myFile.write(Serial.read());
   myFile.flush();
   
   delay(1000);
   Serial.write("ATL1\r");
   delay(2000);
   while(Serial.available())myFile.write(Serial.read());
   myFile.flush();
   
   delay(1000);
   Serial.write("ATSP5\r");   
   delay(2000);
   while(Serial.available())myFile.write(Serial.read());
   myFile.flush();

}
  
void loop()
{       
   delay(2000);
   Serial.write("010C\r");
   Serial.flush();
   unsigned long timer = millis();
   while(millis() - timer < 5000){
     if(Serial.available())myFile.write(Serial.read());
   }
   myFile.flush();

}
```

hayden-t, 2015-04-05:
Results from sketch:
```
Starting Test
OBDUART v1.0

>ATE1
OK

>ATL1
OK

>ATSP5
OK

>010C
BUS
INIT:
...ERROR

>010C
BUS
INIT:
...OK
E2 03 61 8D 

>010C
E2 03 61 8E 

>010C
E2 03 61 8F 

>010C
7E 94 84 F1 10 41 0C 00 00 

>010C
7E 94 00 00 07 E2 03 61 8F 02 84 F1 10 41 0C 00 00 

>010C
7E 94 00 00 07 E2 03 61 8E 00 84 F1 10 41 0C 00 00 

>010C
7E 00 00 07 E2 03 61 8D 7A 94 84 F1 10 41 0C 00 00 

>010C
07 E2 03 61 8F 02 F1 A4 7E 94 84 F1 10 41 0C 00 00 

>010C
E2 03 61 8E 00 81 F1 A4 7E 94 84 F1 10 41 0C 00 00 

>
```

hayden-t, 2015-04-05:
elm327 results:
```
ATZ
ELM327 v1.5

>ATE1
OK

>ATL1
OK

>ATSP5
OK

>010C
BUS INIT: OK
41 0C 000
7F 01 12

>010C
41 0C 000
7F 01 12

>010C
41 0C 000
7F 01 12

>010C
41 0C 000
7F 01 12
```

hayden-t, 2015-04-22:
have now tried on toyota yaris and its fine, so it must be a problem with the logger hardware and protocol ATSP5 in my Citroen, as elm327 works fine.