stanleyhuangyc/Freematics #10: get voltage function in loop without obd

hayden-t, 2015-04-04:
Hi, in commit 4efef6855a0a966cac896e194c1c626e303090e8 you uncommented 
```
    int v = logger.getVoltage();
    logger.logData(PID_BATTERY_VOLTAGE, v);
```

this is in the loop without any checks to see if obd active or not, this it is constantly being run and logged with an incorrect voltage ~ 0.398v

likely it doesnt need to be checked this regularly either

also curious why it needs its own function and isnt collected by logger.logOBDData();

stanleyhuangyc, 2015-04-04:
This will not cause OBD-II connection
      at all.
      
      On 2015/4/4 17:11, hayden-t wrote:
    
    
      Hi, in commit 4efef68 you uncommented 
          int v = logger.getVoltage();
    logger.logData(PID_BATTERY_VOLTAGE, v);

      this is in the loop without any checks to see if obd active or
        not, this it is constantly being run and logged with an
        incorrect voltage ~ 0.398v
      likely it doesnt need to be checked this regularly either
      also curious why it needs its own function and isnt collected
        by logger.logOBDData();
      —
          it on GitHub.

stanleyhuangyc, 2015-04-04:
This will not cause OBD-II connection at all. It's a function provided by the data logger itself.

hayden-t, 2015-04-04:
what i meant is that it is still trying to send ODB requests and interpret them regardless of if the ODB is connected or not.
```
int COBD::getVoltage()
{
    char buf[OBD_RECV_BUF_SIZE];
    write("ATRV\r");
    byte n = receive(buf, 100);
```

stanleyhuangyc, 2015-04-04:
ATRV command is a custom command we implement which does not initiate any OBD-II communication

hayden-t, 2015-04-04:
so where is write("ATRV\r"); sending to ?

stanleyhuangyc, 2015-04-04:
The battery voltage meter and GPS is provided by the data logger hardware itself instead of vehicle ECU.

hayden-t, 2015-04-04:
Ok then why query it if its not connected ? (the battery voltage meter that is)

stanleyhuangyc, 2015-04-04:
No OBD connection does not mean physically disconnected. This typically happens when ignition is off.

hayden-t, 2015-04-04:
So are you saying its sampling the cars battery voltage from obd port pins other than the can bus ?

stanleyhuangyc, 2015-04-04:
Yes I meant that.

hayden-t, 2015-04-04:
ah ok, i get it now. though im thinking it should still have a config.h define to opt out if you dont want it filling up your logs ? like you do for the MEMS sensor. as i imagine it will be logging constantly even if the car is off ?

stanleyhuangyc, 2015-04-04:
The sketch is provided as a demo/testing purpose. So it is good to let something work (LED flashing) even if OBD/GPS/MEMS is absent.

closed 2015-04-04 by stanleyhuangyc

hayden-t, 2015-04-04:
MEMS is always present ? and what do you mean demo/testing purposes ? its the only datalogger sketch that you provide ?

hayden-t, 2015-04-04:
Also it can still be on by default like the it now and the MEMS is too, but there should be an option to turn it off in config.h

hayden-t, 2015-04-04:
actually what would be even better would be to be able to specify the poll frequency as well