stanleyhuangyc/Freematics #14: No support for software serial?

tcados, 2016-02-18:
Is there a reason why software serial won't work? I modified the test code as follows and it still doesn't work:

**GPRSTest.ino:**
added:
#include <SoftwareSerial.h>
SoftwareSerial SIMSer = SoftwareSerial(2, 3); 

**SIM800.h**
// change this to the serial UART which SIM800 is attached to
#define SIM_SERIAL SIMSer

when I try to compile I get the following error:

> In file included from ... SIM800/SIM800.h: In member function 'bool CGPRS_SIM800::available()':
> .... SIM800/SIM800.h:14:20: error: 'SIMSer' was not declared in this scope
>  #define SIM_SERIAL SIMSer
>                     ^
> /Users/troycados/Documents/Arduino/libraries/SIM800/SIM800.h:79:14: note: in expansion of macro 'SIM_SERIAL'
>        return SIM_SERIAL.available(); 
>               ^

Am I doing anything wrong in particular? or does the library just not support software serial?

Thanks!
-T

geologic, 2016-03-04:
The same for me, i need to use library with other digital pins than TX/RX UART

sqwerty13, 2016-06-19:
The same for me, i have this problem too

rox85, 2017-03-21:
In SIM800.hyou could comment the line number 14
`// #define SIM_SERIAL Serial1`

In SIM800.cpp you should insert the following lines at the start of the file:

```
#include <SoftwareSerial.h>
SoftwareSerial SIM_SERIAL(10, 11); // RX, TX
```