stanleyhuangyc/Freematics #16: Downloading sim800 library

Pritha01, 2016-04-04:
Hello everyone!

Can anybody plz help me in downloading "sim800" library. I am unable to find my way.

nodemand, 2016-04-04:
You have to move up a couple of folders and then download the entire project (as a zip) of which the sim800 library is a part.

Pritha01, 2016-04-05:
I have downloaded .ZIP file from [https://github.com/stanleyhuangyc/Freematics](url) named "Freematics-master.zip" but when I am trying to add it to Arduino IDLE it shows "specified folder/zip file does not contain a valid library".

[image]
[image]
[image]

closed 2016-04-07 by Pritha01