stanleyhuangyc/Freematics #17: How Can I send location within an SMS in sim800 GSM module?

Pritha01, 2016-04-08:
Hello everyone!

I am using sim800 module. Using AT commands I am able to send SMS as well as get location of the sim but I want to send the location obtained using AT commands inside an SMS. Can anyone help me with the code? I'm attaching a pic of my GSM module.
[image]
[image]
[image]

nodemand, 2016-04-08:
Are you using the sim800 library and Arduino? If yes, what is the code you have now? If not, what are you using now to send AT commands?

Pritha01, 2016-04-10:
Ya I've downloaded sim800 library from github [https://github.com/stanleyhuangyc/Freematics/tree/master/libraries/SIM800](url) .
I'm using Arduino Uno.
I used following series of AT commands to get location:

AT+SAPBR=3,1,"CONTYPE","GPRS"
AT+SAPBR=3,1,"APN","internet"
AT+SAPBR=2,1
AT+CIPGSMLOC=1,1

I got triangulated location.
I need a way to send this location from within an SMS.
I've read the code provided at [https://forum.arduino.cc/index.php?topic=325325.0](url) but I'm unable to understand how to use it according to my requirements.

To send AT commands using Arduino Uno & to read the response from GSM module I've used the following code:

`#include <SoftwareSerial.h>
 
//SIM800 TX is connected to Arduino D8
#define SIM800_TX_PIN 8
 
//SIM800 RX is connected to Arduino D7
#define SIM800_RX_PIN 7
 
//Create software serial object to communicate with SIM800
SoftwareSerial serialSIM800(SIM800_TX_PIN,SIM800_RX_PIN);
 
void setup() {
  //Begin serial comunication with Arduino and Arduino IDE (Serial Monitor)
  Serial.begin(9600);
  while(!Serial);
   
  //Being serial communication witj Arduino and SIM800
  serialSIM800.begin(9600);
  delay(1000);
   
  Serial.println("Setup Complete!");
}
 
void loop() {
  //Read SIM800 output (if available) and print it in Arduino IDE Serial Monitor
  if(serialSIM800.available()){
    Serial.write(serialSIM800.read());
  }
  //Read Arduino IDE Serial Monitor inputs (if available) and send them to SIM800
  if(Serial.available()){    
    serialSIM800.write(Serial.read());
  }
}`
For the above code I've used Seeduino_GPRS library downloaded from [https://github.com/Seeed-Studio/Seeeduino_GPRS](url)

Pritha01, 2016-04-10:
Oh! I'm afraid! The AT commands that I wrote above were working fine a couple of days ago but I'm getting an error now!
[image]

nodemand, 2016-04-10:
here is my send sms function. i haven't tested it but it should work.

`bool sendSMS(const char number, const char text)
{
    cleanBuffer();
    serialSIM800.println("AT+CMGF=1");
    if ( waitFor("OK", "ERROR") != 1 ) return false;

    cleanBuffer();
    serialSIM800.print("AT+CMGS=\"");
    serialSIM800.print(number);
    serialSIM800.println("\"");
    if ( waitFor(">", "ERROR") != 1 ) return false;

    cleanBuffer();
    serialSIM800.print(text);
    serialSIM800.println((char)26);
    if ( waitFor("+CMGS:", "ERROR") != 1 ) return false;

    return true;
}`

and the cleanBuffer function:

`void cleanBuffer()
{
    delay( 250 );
    while ( serialSIM800.available() > 0) 
    {
        serialSIM800.read();    // Clean the input buffer
        delay(50);
    }
}`

and the waitFor function:

`int8_t waitFor(const char* expected_answer1, const char* expected_answer2)
{
    uint8_t x=0,  answer=0;
    char response[100];
    unsigned long previous;

    memset(response, (char)0, 100);    // Initialize the string

    delay( 250 );

    x = 0;
    previous = millis();

    // this loop waits for the answer
    do{
        // if there are data in the UART input buffer, reads it and checks for the asnwer
        if(serialSIM800.available() > 0){
            response[x] = serialSIM800.read();
            x++;
            // check if the desired answer 1  is in the response of the module
            if (strstr(response, expected_answer1) != NULL)
            {
                answer = 1;
            }
            // check if the desired answer 2 is in the response of the module
            else if (strstr(response, expected_answer2) != NULL)
            {
                answer = 2;
            }
        }
        delay(10);
    }
    // Waits for the asnwer with time out
    while((answer == 0) && ((millis() - previous) < 10000 ));

    return answer;
}`

Good luck!

nodemand, 2016-04-10:
sorry for the bad markdown. i must be doing something wrong. let me try that again.

the send sms function:

    bool sendSMS(const char number, const char text)
    {
        cleanBuffer();
        serialSIM800.println("AT+CMGF=1");
        if ( waitFor("OK", "ERROR") != 1 ) return false;

        cleanBuffer();
        serialSIM800.print("AT+CMGS=\"");
        serialSIM800.print(number);
        serialSIM800.println("\"");
        if ( waitFor(">", "ERROR") != 1 ) return false;

        cleanBuffer();
        serialSIM800.print(text);
        serialSIM800.println((char)26);
        if ( waitFor("+CMGS:", "ERROR") != 1 ) return false;

        return true;
    }

the cleanBuffer function:

    void cleanBuffer()
    {
        delay( 250 );
        while ( serialSIM800.available() > 0) 
        {
            serialSIM800.read();    // Clean the input buffer
            delay(50);
        }
    }

the waitFor function:

    int8_t waitFor(const char* expected_answer1, const char* expected_answer2)
    {
        uint8_t x=0, answer=0;
        char response[100];
        unsigned long previous;
        
        memset(response, (char)0, 100);    // Initialize the string
        
        delay( 250 );
        
        x = 0;
        previous = millis();
        
        // this loop waits for the answer
        do{
            // if there are data in the UART input buffer, reads it and checks for the asnwer
            if(serialSIM800.available() > 0){
                response[x] = serialSIM800.read();
                x++;
                // check if the desired answer 1  is in the response of the module
                if (strstr(response, expected_answer1) != NULL)
                {
                    answer = 1;
                }
                // check if the desired answer 2 is in the response of the module
                else if (strstr(response, expected_answer2) != NULL)
                {
                    answer = 2;
                }
            }
            delay(10);
        }
        // Waits for the asnwer with time out
        while((answer == 0) && ((millis() - previous) < 10000 ));
    
        return answer;
    }

nodemand, 2016-04-11:
And here are some more functions which might help you:

```
bool setupGPRS()
{
    delay( 250 );
    if ( !setConnectionType() ){
      Serial.println("Failed to set connection type to GPRS.");
      return false;
    }
    delay( 250 );
    if ( !setAPN("internet") ){
      Serial.println("Failed to setup APN.");
      return false;
    }
    delay( 250 );
    if ( !startGPRS() ){
      Serial.println("Failed to start GPRS.");
      //return false;
    }
    delay( 250 );
    if ( !hasIP() ){
      Serial.println("Failed to acquire IP address.");
      return false;
    }

    return true;
}

bool setConnectionType()
{
    cleanBuffer();
    serialSIM800.println("AT+SAPBR=3,1,\"Contype\",\"GPRS\"");
    if ( waitFor("OK", "ERROR") != 1 ) return false;
  
    return true;
}

bool setAPN(const char* apn)
{
    cleanBuffer();
    serialSIM800.print("AT+SAPBR=3,1,\"APN\",\"");
    serialSIM800.print(apn);
    serialSIM800.println("\"");
    if ( waitFor("OK", "ERROR") != 1 ) return false;
  
    return true;
}

bool startGPRS()
{
    cleanBuffer();
    serialSIM800.println("AT+SAPBR=1,1");
    if ( waitFor("OK", "ERROR") != 1 ) return false;

    return true;
}

bool hasIP()
{
    cleanBuffer();
    serialSIM800.println("AT+SAPBR=2,1");
    if ( waitFor("0.0.0.0", "OK") != 2 ) return false;

    return true;
}
```

Pritha01, 2016-04-11:
Thank you so much for this! :) I'll try out these and share the result!

Pritha01, 2016-04-12:
Can you please help me with the code to send the location as a value for the parameter "text" in the "sendSms" function?

nodemand, 2016-04-12:
something like this maybe:

variable declaration:

```
char* latitude;
char* longtitude;
char message[160];
```

building the message and sending it:

```
latitude = "0,125463";
longtitude = "0,38757648976453";

memset(message, (char)0, 160);
strcpy(message, "LAT: ");
strcat(message, latitude);
strcat(message, "\r\n");
strcat(message, "LON: ");
strcat(message, longtitude);
strcat(message, "\r\n");
sendSMS( "+31612345678", message );
```

I've changed the sendSMS function from const char variables to char* variables.

```
bool sendSMS(char* number, char* text)
{
    cleanBuffer();
    serialSIM800.println("AT+CMGF=1");
    if ( waitFor("OK", "ERROR") != 1 ) return false;

    cleanBuffer();
    serialSIM800.print("AT+CMGS=\"");
    serialSIM800.print(number);
    serialSIM800.println("\"");
    if ( waitFor(">", "ERROR") != 1 ) return false;

    cleanBuffer();
    serialSIM800.print(text);
    serialSIM800.println((char)26);
    if ( waitFor("+CMGS:", "ERROR") != 1 ) return false;

    return true;
}
```

nodemand, 2016-04-12:
or you could (maybe more easily) change the sendSMS function to sendLocation and go like this:

variable declaration:

```
char* latitude;
char* longtitude;
```

sending the location:

```
latitude = "0,125463";
longtitude = "0,38757648976453";
sendLocation( "+31612345678", latitude, longtitude);
```

the sendLocation function:

```
bool sendLocation(char* number, char* latitude, char* longtitude)
{
    cleanBuffer();
    serialSIM800.println("AT+CMGF=1");
    if ( waitFor("OK", "ERROR") != 1 ) return false;

    cleanBuffer();
    serialSIM800.print("AT+CMGS=\"");
    serialSIM800.print(number);
    serialSIM800.println("\"");
    if ( waitFor(">", "ERROR") != 1 ) return false;

    cleanBuffer();
    serialSIM800.print("LAT: ");
    serialSIM800.println(latitude);
    serialSIM800.print("LON: ");
    serialSIM800.println(longtitude);
    
    serialSIM800.println((char)26);
    if ( waitFor("+CMGS:", "ERROR") != 1 ) return false;

    return true;
}
```

Good Luck!

Pritha01, 2016-04-13:
Thanks a lot! Your sendSMS() function is working properly!
Suppose I use an additional GPRS module to get location in latitude & longitude then how can I store the value obtained separately in these variables:
char* latitude;
char* longtitude;

because only then will I be able to use your new sendLocation() function!

nodemand, 2016-04-13:
Have a look in the source code of the sim800 library files. Especially in the .cpp file. You should be able to figure out how to piece it together...

Pritha01, 2016-04-18:
Hello!

I've found a way to extract location and get Latitude & longitude. I have this array- String nmea[15]
and I have latitude and longitude stored at nmea[2] and nmea[4].

As you guided me, I used following code:

` memset(message, (char)0, 160);
  strcpy(message, "LAT: ");
  strcat(message, nmea[2]);
  strcat(message, "\r\n");
  strcat(message, "LON: ");
  strcat(message, nmea[4]);
  strcat(message, "\r\n");
  sendSMS( "8*********", message );`

and I'm using your modified sendSMS function as you've told above.

`bool sendSMS(char* number, char* text)
{
    cleanBuffer();
    serialSIM800.println("AT+CMGF=1");
    if ( waitFor("OK", "ERROR") != 1 ) return false;

    cleanBuffer();
    serialSIM800.print("AT+CMGS=\"");
    serialSIM800.print(number);
    serialSIM800.println("\"");
    if ( waitFor(">", "ERROR") != 1 ) return false;

    cleanBuffer();
    serialSIM800.print(text);
    serialSIM800.println((char)26);
    if ( waitFor("+CMGS:", "ERROR") != 1 ) return false;

    return true;
}`

I don't know anything about memset(). When I compiled the code, I got following error:

> 'message' was not declared in this scope

So I declared it like String message as I did not know what memset() is doing.

When I did so, I received following error on compiling:

> cannot convert 'String' to 'void*' for argument '1' to 'void* memset(void*, int, size_t)'

Please help me!

Pritha01, 2016-04-18:
Sorry for the bad markdown, but I'm unable to correct it! :(

stysonchanvilla, 2017-05-23:
@Pritha01 Could you please post the entire program? It would be of great help. Thanks!

devildivesh21, 2017-10-07:
ya @Pritha01 @nodemand please post the entire code to get the location through sms it will be of great help thanks

Pritha01, 2017-10-07:
My problem has been resolved thank u guys so much. Sorry for informing late.

JosephAdamite, 2017-10-07:
hello @Pritha01 please can you post the code it would be helpful for others

SagarUdayan, 2017-10-13:
please @Pritha01  can you post this code.

WA4OSH, 2017-10-13:
You will need to write a script that reads the NMEA text (usually at 4800
Baud serial over USB) from a GPS and parse it.  You will need to find the
NMEA sentence that contains the GPS location.  There a couple of dozen more
for other purposes for things like date/time, satellite locations, etc.
http://www.gpsinformation.org/dale/nmea.htm

You will want to find the location string.

 $GPGGA,123519,4807.038,N,01131.000,E,1,08,0.9,545.4,M,46.9,M,,*47

Where:
     GGA          Global Positioning System Fix Data
     123519       Fix taken at 12:35:19 UTC
     4807.038,N   Latitude 48 deg 07.038' N
     01131.000,E  Longitude 11 deg 31.000' E
     1            Fix quality: 0 = invalid
                               1 = GPS fix (SPS)
                               2 = DGPS fix
                               3 = PPS fix
			       4 = Real Time Kinematic
			       5 = Float RTK
                               6 = estimated (dead reckoning) (2.3 feature)
			       7 = Manual input mode
			       8 = Simulation mode
     08           Number of satellites being tracked
     0.9          Horizontal dilution of position
     545.4,M      Altitude, Meters, above mean sea level
     46.9,M       Height of geoid (mean sea level) above WGS84
                      ellipsoid
     (empty field) time in seconds since last DGPS update
     (empty field) DGPS station ID number
     *47          the checksum data, always begins with *

And then build an AT command you send your GSM module.

--Konrad, WA4OSH

Pritha01, 2017-10-15:
@SagarUdayan which part of the code you need help with? :)

Pritha01, 2017-10-15:
I visited http://davidjwatts.com/ and mailed my query to David J Watts. His code helped me to extract the required information from NMEA text. :)

SagarUdayan, 2017-10-15:
actually i have sim 800l and i want to send my location to a sql server and
i am totally new to all these stuffs please can you guide me

khossanr, 2017-10-29:
Thanks for sharing... i need full code like.. when i send a sms like "status" to gsm 800L module then got the location on my cell phone with google map link,anybody here help me please.

Pritha01, 2017-10-29:
@SagarUdayan  I can help you with the first part of the code, to receive the location from gsm module and send the location within SMS but for sending it to an sql server, you need to research a lot since I haven't tried it.

mxfxm00, 2017-12-26:
@Pritha01 did u make declaration for longitude and latitude? and can it be updated if i use gps module?

padiee, 2017-12-26:
Pritha that sites for code is not opening can you send me that via email to padiee755@gmail.com

Pritha01, 2017-12-27:
https://youtu.be/bgOZLgaLa0g
 sorry guys due to an issue I'm unable to help you with code right now. Plz follow this YouTube link of David J Watts. I am sure it will solve most of your problems related with latitude and longitude.

closed 2017-12-27 by Pritha01

padiee, 2017-12-27:
Thank you so much
If you are having any data regarding this you can send it me please my
contact number is 7057557851
That how to send gps location via sms to phone number by aurdino and gsm
module

ellisvalentiner, 2017-12-27:
@padiee This is a public form, not a private support chat. You probably shouldn’t post your phone number.

padiee, 2017-12-28:
> 354057220>,

Ohk can you send me the any information

reddy-jahnavi, 2018-01-28:
@padiee I too need information regarding the same.If you have any updates please tell me through Gmail.My mail id is 
reddyjahnavi22@gmail.com

Arpan321, 2018-01-31:
hiii...
I'm doing a project using sim800A(simcom) and arduino uno ... I need help in programming  to send & receive sms & location 

please help me as soon as possible
my email id - arpanmarch28@gmail.com

thank you...

[image]

rafaelnfs, 2018-05-10:
/////////////////////////////////////////////////////////////////////////
                              THIS WORKS FOR MY
///////////////////////////////////////////////////////////////////////

AT+CGATT =1 // to attach GPRS.

AT+SAPBR =3,1,”CONTYPE”,”GPRS” //activate bearer profile.

AT+SAPBR =3,1,”APN”,”RCMNET”

AT+SAPBR=1,1

AT+SAPBR=2,1

AT+CIPGSMLOC=1,1 //to get gsm location, time and date.

AT+CIPGSMLOC=2,1 //to get gsm time and date

AT+SAPBR =0,1 //to deactivate bearer profile.

FatimaZohraKouddad, 2021-03-07:
@Pritha01 ; hi ; plase help me, you can send me the full code it's possible ?
gmail: kouddadfatimazohra@gmail.com
 and thank u :)

Pritha01, 2021-03-08:
Hi Fatima,

I wrote the code as a part of my engineering final project, around 8 years
ago. That code was not perfect & I have lost touch with it.

Please let me know what exactly are you looking for & what is your purpose.
Will try to look for it & send you.

Thanks & Regards,
Pritha