stanleyhuangyc/Freematics #18: Freematics One Datalogger: GPS Doesn't Attempt to Reconnect

JonnoFTW, 2016-08-05:
It seems that if the logger can't find GPS when it starts up, it gives up forever. It would be nice if it tried again, I've implemented it and it looks like this:

In `loop()`:

```
#if USE_GPS
  if (one.state & STATE_GPS_FOUND) {
    one.logGPSData();
  } else {
    one.connectGPS();
  }
```

And in `ONE` class:
``` 
    void connectGPS() {
      static unsigned long GPS_LAST_TIME = 0;
      if (!GPS_LAST_TIME || millis() > GPS_LAST_TIME + 10000) {
        delay(100);
        GPS_LAST_TIME = millis();
        SerialRF.print("GPS,");
        if (initGPS(GPS_SERIAL_BAUDRATE)) {
          state |= STATE_GPS_FOUND;
          SerialRF.println("OK");
        } else {
          SerialRF.println("NO");
        }
      }
    }
```

mikebolt, 2016-10-26:
Thanks for this code. It was very helpful.

I think I have found a better solution. The initGPS function sends three SPI commands:

1. ATGPSON
2. ATBR2<rate>
3. ATGRR

In my testing, the first two commands never failed. The ATGRR command is only sent to determine whether the GPS is currently connected. It might fail the first time then succeed after that, without even re-sending the first two commands. It is independent. Once you send the first two commands, it is ready to go, and ATGRR will return raw data whenever it can, and "$GPS NO DATA" or something whenever it can't.

This suggests that instead of calling initGPS, just send ATGRR every loop. If it works, log the data. If not, don't log any GPS data that loop. This should make the logger more efficient, because re-sending ATGPSON and ATBR2 takes some time.

You will either need to incorporate TinyGPS, or figure out how to use getGPSData in a similar way.

JonnoFTW, 2016-10-26:
How would this handle cases where the device loses all connection to the GPS satellites (such as going through a tunnel)?  Which command specifically tells the GPS module to reconnect rather than just turning it on?

Thanks for the suggestion, I'll have a look at incorporating your changes.

bestkanthed, 2017-06-05:
My device never returns any data.
I did.
ATGPSON
ATBR2115200
ATGRR in loop
I roam around with it in "open space" but still NO DATA.
Help!