stanleyhuangyc/Freematics #20: GPS values are not recorded at full precision in the datalogger sketch

mikebolt, 2016-10-03:
The values for GPS speed and GPS heading are not being recorded at their full precision. Speed is rounded to whole kilometers per hour, yet it is available from the raw NMEA GNRMC message in knots with a precision of three digits after the decimal place. Likewise, heading is available in degrees with a precision of two digits after the decimal place, but it is rounded to whole degrees.

I am not sure where the $GPS messages are coming from, but I speculate that they are created from the raw NMEA messages on the STM32. At least, I couldn't find any place where the $GPS messages are created within this project.

It should be possible to use TinyGPS in the datalogger sketch just like the telelogger sketch. TinyGPS appears to handle the decimal precision of values in the NMEA data. I plan to work on integrating TinyGPS into the datalogger sketch on my own, but I thought this was worth posting here anyway.

stanleyhuangyc, 2016-10-15:
From my past experience, I seldom met with occasion that requires very high precision of GPS speed as the speed is estimated from lat/lon change. That's why I chose int type for the GPS speed.

mikebolt, 2016-10-15:
GPS speed is estimated with the help of Doppler shift, not just the track.
I have actually plotted the GPS speed from this device using a sample of
the raw NMEA messages, and it looks great. I can upload that plot later for
reference.

Besides, we need the most accurate data we can get.

dogmatic69, 2016-10-15:
Speed should not be an integer, its provided as decimal from the device (NEMA spec).

mikebolt, 2016-10-26:
I have successfully rewritten the datalogger sketch so that it uses TinyGPS to parse the values returned by getGPSRawData. I think this is the only way to fix the issue. I am willing to incorporate this modification into the FreematicsONE library. Let me know if you think it is worth the effort.