stanleyhuangyc/Freematics #21: Too many DAT#####.CSV files are created by the datalogger

mikebolt, 2016-10-03:
The datalogger sketch creates an extra DAT file each time it creates a DAT file. The result is that it always creates an even number of DAT files, and all of the odd-numbered DAT files are completely empty.

The following code is probably relevant:

      } else {
        // no GPS connected 
        int index = one.openFile(0);
        if (one.openFile(0) != 0) { // TODO: fix duplicate DATA file bug.
                  // when one.openFile is only called once, no data is written???
          one.state |= STATE_FILE_READY;
          SerialRF.print("FILE ");
          SerialRF.println(index);
        } else {
            SerialRF.println("File error");
        }
      }

The problem appears to be that `one.openFile(0)` is called twice: once to save the result into `index`, and another time to use the result in the if statement. I tried replacing the second call in the comparison expression with the value of `index`, but that didn't work for some reason. I think that it resulted in no data being recorded at all.

stanleyhuangyc, 2016-10-15:
I have observed such phenomenon. Just note that when you open the serial monitor, the atmega328p will reset and that will cause a new file to be created.

mikebolt, 2016-10-15:
OK. That's good to know.

Duplicate files are always created, even when the device is powered through
OBD and not connected through USB. Data is always written to even numbered
data files, and odd numbered files are always empty.

closed 2016-10-21 by stanleyhuangyc