stanleyhuangyc/Freematics #26: unstable SIM800L EVB

erloppy, 2017-01-10:
Hello
I am testing test code with an arduino mega2560 and a module gsm SIM800L EVB, but the result is very unstable, sometimes it gets many connections and sometimes it does not, it stops or it gives me connection error continuously.

These doing a project that consists of sending samples from a sensor to a bbdd and I thought that the SIM800L module was a good idea, but I have doubts, is anyone having these problems?

Thank you.

And thanks for the library.

mbenitez01, 2017-03-05:
I managed to send 100 urls contianing "sensor information" to my remote server... 100 GET request..one  each 15 seconds... all returned 200 (success)....after that, I decided to to a more real scenario and changed inteval to 15 MINUTES...left the process running and went to bed..... I gess my pc run into hibernation satus, and the process stooped suddenly..... when I woke in the mornig...I HAVE NEVER..NEVER..NEVER been able to send a GET again..not a single one!!
wrote to SimCom enigineers who basically told me the 800L is a piece of crap...they strongly discouraged me to use it in production...they said this was a cheap toy just for test in the china market......so ..if their engineers talk like this.... we should better forget this module, I guess...

bkkrunal, 2017-03-21:
Hi @mbenitez01 
Your comment on SIM800L makes me worry.
I have used SIM800 (not SIM80L) in production and now I am worried if its the same case with SIM800.
Did yo find any other better option for industrial standard GSM modules?

Thanks in advance.

mbenitez01, 2017-03-21:
Hi..

For all I have read about SIM800x, I came to the conclusion that 800L is
some kind of defective brother of the family...and the manufacturer decided
to leave that 'child' only for domestic use, where they surely would avoid
demands. Also, this problem caused module price drop to near zero. However,
this extremelly low price was
seen by unscrupulous  marketers as an excelent oportunity to make huge
economic difference and despite the module flaws, decided to market it all
around the world.

The healtly brothers, (800C, 800H and 800) work fine as a far as I know.

Good luck with your development..may yoy please keep me updated how it
goes?
(btw..do you send gets/posts, or just plain sms?)

regards

Marcelo Benitez
15-3660-0730
4661-9462
pedromarcelobenitez(skype)
soluciones.16mb.com

mbenitez01, 2017-03-21:
I have not found a 'champ'..right now I have started testing of al.thinker
A7 (extrmely poor) and n old SIM340...
I have to solder components on a quectel m95..but at this point I'm kind of
dissapointed to start new development..

Marcelo Benitez
15-3660-0730
4661-9462
pedromarcelobenitez(skype)
soluciones.16mb.com

bkkrunal, 2017-03-22:
Hi,

Thank You very much for the reply.
So that is the reason for why SIM800L can not be officially seen on their website.

I have been developing with SIM800C from last month and making GET/POST requests.
The main problem I am facing now is that if I don't make any request for more than 10-15 minutes, the module will stop working.
I am trying to reset the module programatically when it happens but still can't re-establish communication.
The module starts working with the whole system reset so I am trying to figure out if its module problem or a bug in the code.

I would love to keep you updated on the same.

Thanks again for your help & good wishes.

Krunal
bkkrunal2411@gmail.com
krunal.bhongade (skype)

mbenitez01, 2017-03-22:
Yes..I have faced that problem...what you can do is execute an ATI, or
AT+COPS=? or even a plain AT..to keep module alive.....to maintain
connection, you can execute AT+CIPPING...or AT+SAPBR=2,1....

Marcelo Benitez
15-3660-0730
4661-9462
pedromarcelobenitez(skype)
soluciones.16mb.com

erloppy, 2017-03-22:
Hello, it is very interesting that you comment.

I am developing a device that writes values from a sensor in a bbdd using an 800L and I am having many stability problems.

The reason for using the 800L is by price and by energy consumption, my device will be powered by batteries and I want it to last two years and this is very complicated with the current GSM modules because they consume a lot, and the 800L has a great consumption .

Do you know of any other GSM module that has consumption similar to 800L and is more stable ?.

Thank you.

mbenitez01, 2017-03-22:
I'm very dissapointed with 3 or 4 modules I have tested so far.....

I bought an AI-thinker A7 a couple of months ago....I started to test it
two weeks ago and I have to say this is barely more than a stone....I was
not able to connect to the GPRS network...! But I have to give credit to
the AI-thinker guys...after all, they are the parents of 8266  monster..and
their plan is to upgrade firmware to A7..so you can 'drop' a LUA on it....
they do not provide an estimated date..hopefuly it happens during 2017..at
that point I guess we'll have an stable gsm module.

Marcelo Benitez
15-3660-0730
4661-9462
pedromarcelobenitez(skype)
soluciones.16mb.com

bkkrunal, 2017-03-22:
Hi @mbenitez01,
Thanks for the suggestions.

 I tried to execute AT instructions after every minute but still the same problem.
But, now I have found a work around that I am making dummy GET request every once in 2 minutes, and now its been working good.
Now, the only problem I see that I haven't yet found any way to reset the module.

@erloppy, SIM800L is as equal as SIM800C in cost but I am not sure about power consumption.

I will update my test results here.

Thanks.

mbenitez01, 2017-03-23:
you left me thinking with your email...
and I'm afraid I have bad news for you...
any RF transceiver (434Mhz, 900Mhz, BT, 2,4G , wifi....you name it) needs
aprox 100 mA during transmition...(some use 70 or 80 milliamps, but it's no
big difference)

now...each time the module has to transmit, it takes aprox 30 seconds (half
a minute) to complete the cycle...

some maths:

TXburstenergy=100 mA* 1/120 h                (1/120h is 30 seconds)

TXburstenergy=1mAh (aprox)

now.....if you transmit once a minute, you will need 1440 mAh per day  (in
one day you have 1440 minutes=60x24)

so..let's supouse you have a batery pack yielding 5000mAh....this will give
you energy just for 3,5 days!

(5000/1440= 3,5 aprox)

Let´s say I exaggerated with my numbers....let's change TX period to one
each 15 minutes.....

That gives a batery autonomy of 50 days aprox (3,5x15)

Once again, if you want to halve power consume...fianlly you'll get 100
days being very optimistic...

I guess a solar panel will be a good solution....

Hopefuly this helps.

Marcelo Benitez
15-3660-0730
4661-9462
pedromarcelobenitez(skype)
soluciones.16mb.com