stanleyhuangyc/Freematics #33: Is there a way to programmatically control the LED on the Freematics ONE?

timurrrr, 2017-04-25:
I want to use the LED to show whether logging is in progress or not; and also to blink on certain events, such as errors.

However, I can't achieve that using LED_BUILTIN. It seems like the LED is wired to the serial TX?

corradocampisano, 2021-09-26:
> I want to use the LED to show whether logging is in progress or not; and also to blink on certain events, such as errors.
> 
> However, I can't achieve that using LED_BUILTIN. It seems like the LED is wired to the serial TX?

where did you get that "LED_BUILTIN" constant?

I couldn't find it in the library files, where is it?

lalten, 2021-09-26:
https://github.com/arduino/ArduinoCore-avr/blob/9f8d27f09f3bbd1da1374b5549a82bda55d45d44/variants/standard/pins_arduino.h#L54