stanleyhuangyc/Freematics #43: avrdude: stk500_recv(): programmer is not responding

OlexandrStepanov, 2018-02-23:
Hello,

I have Freematics V4 device, with BLE and SIM5360E on board.
I have managed to upload sketches on it before.

I didn't upload on it for a while (~1 month), and there were several updates on my Mac:
 * I have updated to macOS High Sierra (10.13.3) from macOS Sierra.
 * I have updated Java to ver.8 build 161 from ver.8 build 151

I can't upload Sketches to the board anymore.
The error I see in Arduino IDE is next:
```
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
```
I have already tried to update USB driver from the latest version from [here](https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers).
Updated all boards and libraries in the Arduino itself.

When I try to Burn bootloader, it says next:
```
avrdude: Error: Could not find USBtiny device (0x2341/0x49)
```

Any help is appreciated.

stanleyhuangyc, 2018-02-28:
The MCU is atmega328p instead of attiny.

OlexandrStepanov, 2018-02-28:
@stanleyhuangyc,
Sorry, I didn't get your point. What I'm doing wrong, could you elaborate on this, please?

OlexandrStepanov, 2018-02-28:
@stanleyhuangyc,
Do you mean I have to use another Arduino Uno to burn bootloader on this Freematics One device?
If so, I have found a lot of tutorials in internet, but arduino devices are quite different there, and I'm not sure which tutorial to follow. Could you suggest one, please?

OlexandrStepanov, 2018-03-02:
Question was raised on the forum as well:
https://freematics.com/forum/viewtopic.php?f=11&t=2693

closed 2018-04-19 by OlexandrStepanov