stanleyhuangyc/Freematics #46: Why my OBD-II UART Adapter V2.1 says haven't MEMS?

ivanzhenlms, 2018-06-26:
Function memsInit always return false and I don't know why.

```
bool COBD::memsInit(bool fusion)
{
	char buf[16];
	if (sendCommand("ATTEMP\r", buf, sizeof(buf)) <= 0 || strchr(buf, '?'))
		return false;
	if (fusion) {
		m_fusion = true;
		return sendCommand("ATQU1\r", buf, sizeof(buf));
	} else {
		if (m_fusion) {
			m_fusion = false;
			sendCommand("ATQU0\r", buf, sizeof(buf));
		}
		return true;
	}
}
```

I checked and it always return first return in code and doesn't perform rest.

```
if (sendCommand("ATTEMP\r", buf, sizeof(buf)) <= 0 || strchr(buf, '?'))
		return false;
```

I need some help please. Thanks.

stanleyhuangyc, 2018-07-20:
Can you make sure what you have is a V2.1 adapter?

ivanzhenlms, 2018-07-20:
Yes, I am sure.

ivanzhenlms, 2018-07-20:
@stanleyhuangyc Yes, I am sure.

ivanzhenlms, 2018-07-21:
There was a packaging issue and I received OBD-II UART Adapter V1 instead of V2.1.

closed 2018-07-21 by ivanzhenlms