stanleyhuangyc/Freematics #47: sd card mounting issue

laurentvm, 2018-06-28:
With brand new freematics, there some software installed on those. When you run it, there are sometimes issue when mounting the card:

```
--- Miniterm on COM7  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP32 240MHz 4MB Flash
Power-saving disabled
MEMS...9-DOF OK
ACC BIAS...-0.06/-0.00/-1.00
SD Card:15732MB
OBD...
```
Works ok. You relaunch again, no card found.

```
--- Miniterm on COM7  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP32 240MHz 4MB Flash
Power-saving disabled
MEMS...9-DOF OK
ACC BIAS...-0.08/-0.00/-1.09
No SD Card
OBD...
```

Would it be possible to get the code running on new devices? Why is it failing while the card is still in.

laurentvm, 2018-07-18:
This issue is still alive. Stanley would you have an idea.

stanleyhuangyc, 2018-07-20:
How did you "relaunch"?

AndreaBorgia-Abo, 2019-04-27:
I am having a similar, perhaps identical issue with the current version:

```
ESP32 240MHz 4MB Flash
MEMS...OK
ACC Bias:0.00/-0.23/0.77
SD...[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
```

I can understand FS corruption when the firmware crashes. However, this happens also when I flash a new firmware: upon restart, I get this error. I have to remove the card, reformat and restart yet again.

AndreaBorgia-Abo, 2019-05-01:
In my case there is a [solution](https://github.com/espressif/arduino-esp32/issues/2171#issuecomment-446627299) at least.

lalten, 2019-05-04:
I think my SD card had a gpt on it instead of msdos partition table. not sure what I changed in the end, sorry...

AndreaBorgia-Abo, 2019-05-04:
> I think my SD card had a gpt on it instead of msdos partition table. not sure what I changed in the end, sorry...
Oh, ok, similar outcome, different reason.

I guess I'll have to implement the workaround, I'm being hit by this issue about half the times I flash a new firmware, quite annoying.