stanleyhuangyc/Freematics #49: Implement support PID

jugendhacker, 2018-07-22:
Is there a way that you could implement the [0x00 PID](https://en.wikipedia.org/wiki/OBD-II_PIDs#Mode_1_PID_00) so it will be possible to request wich pids are supported by the car?

jugendhacker, 2018-12-30:
@alexandrehsd as you can see he isn't responding to any of the Issues nor PRs... I also wrote him an email but he never responded, sorry!

alexandrehsd, 2019-01-03:
That's a pity... So how did you solve this problem? did you implemented something in order to get the supported PIDs?

jugendhacker, 2019-01-04:
I just bought an ELM 327 and a NanoPI and now building it from scratch myself...

alexandrehsd, 2019-01-04:
Oh, I got it. Good luck! I'll try another approach and if I succeed I'll tell you here

AndreaBorgia-Abo, 2019-05-12:
Reading libraries/FreematicsPlus/FreematicsOBD.cpp at stage 3 gives me the idea that the init function will try all PIDs and take stock of what is supported and what isn't, rather than ask.
Have you tried printing the values from pidmap?

Ajaygirish1, 2020-02-28:
You can add an additional PID by following the below steps:
1. Open OBD.h file and add the PID name and hex code to the existing list.
2. Perform a search for an already exiting PID name in the project file to find all places where the PIDS are listed
3. Add your PID to all the places found in the search result.

I compiled the code with the changed PID and found no errors. 
I'm yet to check the working in vehicle.

Hope this helps and is not too late!!