stanleyhuangyc/Freematics #56: Traccar Client feature request

5of9, 2018-08-13:
Hi,
I've a feature request for the traccar client software for Wifi:
I would like to have a intervall saved logging in the internal memory and transmission when wifi is again available. And because the Freematics One+ has OBD it should be used to read the mileage from the which is exact and not wrong like the gps distance. If available the VIN should be used as the device id and also voltage and sats should be transmitted. With this possibilitys I would think about buying a few dozen of it for my company. Thank you

laurentvm, 2018-09-07:
Hi, 

You can quite easily do that with a buffer in RAM or SPIFFS.
I've done that with a queue, mutex and task.

Basiscally, you have 2 tasks: one pushing to queue from sensors and 1 pulling from queue to server.
`
// Callback methods prototypes
void sensorCallBack();
void serverCallBack();

//Tasks
Task t2(100, TASK_FOREVER, &sensorCallBack); // every 100 ms = 0.1
Task t3(3000, TASK_FOREVER, &serverCallBack); //every 5000 ms = 5.0
`

`
for (int i = 0; i < qSize; ++i){
        if (xSemaphoreTake(queueMutex, pdMS_TO_TICKS(SD_IO_WAIT_FOR_SPI)) == pdTRUE){
            String stringURL(queue.pop());
            xSemaphoreGive(queueMutex);
            Serial.print(stringURL);
            if (!net.httpSend(HTTP_GET, stringURL, true)) {
                Serial.println("Error sending data");
                net.httpClose();
            }
            
            // output server response
            if (net.state() == HTTP_CONNECTED) {
                // waiting for server response while still decoding NMEA
                for (uint32_t t = millis(); !net.checkRecvEvent() && millis() - t < 3000; ) {
                }

                char *response;
                int bytes = net.httpReceive(&response);
                if (bytes > 0) {
                    Serial.println(response);
                }
            }
        }
    }
`

laurentvm, 2018-09-07:
You can have a look at telelogger source code, there is a storage approach there