stanleyhuangyc/Freematics #59: Firmware V5 Datalogger web api crashes

BevDan, 2018-08-14:
I have tried to use the latest versions of Firmware V5 Datalogger in the freematics one device, I had an older version that was pretty much working, however I have upgraded to the latest version and now have several issues:
Firstly if I try to get live data without an active obd connection (Not having turned the keys on so the computer is still shutdown, however I have 12v supply so not getting brownouts) and I issue the live request http://192.168.4.1/api/live I get:

`[8193] Request path: api/live
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x40101e30  PS      : 0x00060430  A0      : 0x80107a89  A1      : 0x3ffd1520  
A2      : 0x3ffe6295  A3      : 0x00003df7  A4      : 0x00000154  A5      : 0x00000000  
A6      : 0x0000007d  A7      : 0x0001ae74  A8      : 0x80101e1c  A9      : 0x3ffd1460  
A10     : 0x00000153  A11     : 0x3ffe63e8  A12     : 0x3f408bf0  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000022  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x40101e30:0x3ffd1520 0x40107a86:0x3ffd1580 0x401084b8:0x3ffd15f0 0x40109f75:0x3ffd1620 0x4010481c:0x3ffd1680 0x401025e0:0x3ffd16a0 0x40104acf:0x3ffd1750

Rebooting...
ESP32 240MHz 4MB Flash
MEMS...OK
ACC Bias:0.04/0.02/-1.03
SD...15079MB
HTTP Server...AP:192.168.4.1 OK
OBD...NO
Standby
ACC Bias:0.04/0.02/-1.03
`
And now when I try to get the data over the network I get either blank webpage in Internet Explorer, or and error about unrecognised format when I do the request from within Visual Basic for Applications (MS Access).

I also had to reinstall the Arduino builder to get this version to compile and build properly.

The freematics web page indicates that the most recent version of Arduino builder should be 1.0.12 but the most offered link at sourceforge points to version 1.0.8. I just keep reinstalling version 1.0.11 when I have build troubles

AndreaBorgia-Abo, 2018-08-19:
Dan, regarding the blank page, could you have a look at [issue #52](https://github.com/stanleyhuangyc/Freematics/issues/52) ? It might be related.

As for "live" not working without an OBD connection, I might agree that crashing is not a good thing, but how would it work in practice? Best option here, adding a check for STATE_OBD_READY and then returning with an empty page / error message.

gyroly, 2018-09-04:
Any news on this?

AndreaBorgia-Abo, 2019-03-23:
Crashes in handlerLiveData when trying to print GPS information, which might not yet be available.