stanleyhuangyc/Freematics #77: Under-voltage warning power down error at sim5360test execution

marrakchino, 2019-02-18:
Hi, 

I am trying the `(./firmware_v5/sim5360test`) program on my brand new Freematics One+ but have the following problem at SIM5360 (`net`) setup. Following is the log I read from the board:

```
OK

START
ATI
Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM5360E
Revision: SIM5360E_V3.5
IMEI: 86***********09
Connecting network.........
UNDER-VOLTAGE WARNNING POWER DOWN

starting power off the module!

SMS DONE

+CPSI: NO SERVICE,Powering Off

OK

NO

```

Please note that I power the board with a Micro USB-USB (computer) cable. 

I thank you in advance for your help.

laurentvm, 2019-02-18:
You need 12V power. It does not work only with USB 5v

marrakchino, 2019-02-18:
Hi @laurentvm thanks for the quick reply.
I can't find this information in the product page (https://freematics.com/products/freematics-one-plus/) , could you please provide me with more details?

laurentvm, 2019-02-18:
I do not have details but had the same issue. If you power it through ODB on the right pin or with a OBD power adapter, you will not see this message anymore and the SIM module will work. It's maybe not related to 5V but the amperage ( I do not know).

You can have a look on google about SIM5360 Power supply.

laurentvm, 2019-02-18:
I was powering the PIN 4,5 and 16 with 12V.

https://components101.com/sites/default/files/component_pin/OBD2-Connector-Pinout.png

marrakchino, 2019-02-18:
@laurentvm thanks again I thought of this solution as well but I hesitated to try it. This particular problem is solved but I think the project lacks documentation.

closed 2019-02-18 by marrakchino