stanleyhuangyc/Freematics #83: Issues with HTTP(S) POST request

marrakchino, 2019-02-26:
Hi,

I am using `sim5360` sketch to send a HTTPS POST request but have the following issues.

* When connecting to port 80 with `AT+CHTTPSOPSE=$URL,80,1`, I get the following 302 response:
```
HTTP/1.1 302 Found
Server: Cowboy
Connection: keep-alive
X-Powered-By: Express
Location: $URL
Vary: Accept
Content-Type: text/plain; charset=utf-8
Content-Length: 74
Date: Tue, 26 Feb 2019 14:59:44 GMT
Via: 1.1 vegur

Found. Redirecting to https://<same_server_address>
```

* When connecting to port 443 with `AT+CHTTPSOPSE=$URL,443,2`, I fail to connect.

Does anyone have any insights?
Best regards.

closed 2019-06-04 by stanleyhuangyc

dogusural, 2020-03-13:
I also have the same problem, have you managed to solve it ?

marrakchino, 2020-03-13:
Hi Dogus,

Unfortunately I couldn't. And Stanley didn't seem to consider this a
noteworthy issue.
Good luck solving it...

dogusural, 2020-03-13:
Hi marrakchino ,

Thanks for letting me know. I will post under this post if i manage to solve it.