stanleyhuangyc/Freematics #89: Cant Upload freematics Plus code

Fabrice-Beya, 2019-05-02:
Each time i try to upload i get folowing error:

Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
*** [upload] Error 2

Can someone please help me solve this issue?

oliwilson, 2019-05-16:
I had this issue also, try downloading an earlier version of the freematics builder. One of them has the option to change the comm port from Comm3 to Comm12. Use Comm12 when uploading. Alternatively use the ardunio IDE and switch between the ports to find the correct port.

hurukku098, 2021-02-18:
I had same issue too,

I tried 
1. connect DLC port to supply power
2. download earlier version of freematics builder
3. change usb cable
4. use platformIO, arduinoIDE instead freematics builder
5. change com port number

and finally solved it by using another usb port in may laptop