stanleyhuangyc/Freematics #91: datalogger, missing timestamp

AndreaBorgia-Abo, 2019-05-05:
Until the moment the GPS comes online and provides date and time, at the moment there is no timestamp for the log entries in the file.
The first entry is 10D (speed) and the last is 24 (battery voltage) but no indication of time or any way to correlate it with later GPS date/time information.

stanleyhuangyc, 2019-05-23:
Use timestamp

AndreaBorgia-Abo, 2019-05-25:
@stanleyhuangyc I got a notification for a comment on this issue but I can't find anything. It said "use timestamp". Well, this is what my PR wanted to do but you rejected it. Was there anything wrong with it?

stanleyhuangyc, 2019-08-08:
Timestamp is actually stored as a data set with PID=0.

closed 2019-08-08 by stanleyhuangyc

AndreaBorgia-Abo, 2019-08-08:
Datalogger is _not_ logging to CSV files: that is Telelogger you have in mind. This bug is about Datalogger.