stanleyhuangyc/Freematics #93: /api/log/[file #] returns empty file

AndreaBorgia-Abo, 2019-05-05:
As of today (commit 15a4a960c734c541d71a1ff344eb31ceeb8f8f59), the log API in firmware_v5/datalogger returns an empty file.

There are a few files on my SD card, picking any of those will show on the serial port that the webserver handler reads so and so many bytes (and this matches the list provided by /api/list and the contents of the card) but the client will always get a zero byte file.

Tested with Firefox, wget and plain telnet.
Tested at home with USB power and in the car with the engine running ("live" API returns reasonable data).

BevDan, 2019-07-04:
Can confirm that this issue still exists, along with the lack of timestamp PID 0 when using the latest version of the v5 datalogger. Thankfully I kept a copy from a year ago, to return to, so that most of the functions work. It seems so hard to get any answers to problems with these loggers.

AndreaBorgia-Abo, 2019-07-04:
@BevDan please see [this PR](https://github.com/stanleyhuangyc/Freematics/pull/92) for the timestamp and let me know how it goes (you may reply directly under the PR, rather than here, to keep it clean!).

I have had no time to retest the empty file issue, it seems the datalogger is not a priority for @stanleyhuangyc

BevDan, 2019-07-04:
I will delay loading the latest version onto the logger until the empty file issue is also fixed, your fix looks very simple and straight forward to implement. The timestamps would not be such a big issue for me as I would just add my own index in lieu of the timestamp, but I need to be able to get the data from the logger over the wireless connection

closed 2021-04-08 by AndreaBorgia-Abo