stanleyhuangyc/Freematics #97: Last Datalogger update

Abelardo94, 2019-07-04:
I've been working with Freematics One Plus in the last year, and updating to the last version I've seen some things that have improved but some things that I don't like.
- The timestamp has disappeared and that is really important because sometimes the GPS doesn't work.
- The sampling frequency is lower and a lot of data get lost, I mean, before I use to get like all PIDs data each 200 ms but now I only can get each 700 or 800 ms 

Please if you could help me how to solve this. 
Best regards.

AndreaBorgia-Abo, 2019-07-05:
Glad to see I'm not the only one :)
https://github.com/stanleyhuangyc/Freematics/issues/91

I don't have a solution for the sampling issue, though. I had a feeling something was off but I had no time to test it.

stanleyhuangyc, 2019-08-08:
Timestamp is actually stored as a data set with PID=0.

stanleyhuangyc, 2019-08-08:
Sampling frequency varies with different ECUs. Are you testing on the same vehicle?

AndreaBorgia-Abo, 2019-08-08:
@stanleyhuangyc I only have one car and, no, Datalogger is _not_ logging to CSV files: that is Telelogger you have in mind. This bug is about Datalogger.

Abelardo94, 2019-08-08:
I'm working calculating a timestamp because i couldn't find it but I'm gonna prove with PID(0)
The new problem I have is that datalogger works just the first time is connected, so the problem is that I have to disconnect it and connect again, but I'm working with a lot of vehicles so I can't do that, please do you have some solution?

AndreaBorgia-Abo, 2019-08-08:
@Abelardo94 apply this [PR](https://github.com/stanleyhuangyc/Freematics/pull/92) and you'll have the timestamp back. No ideas for the disconnect issue, sorry :(

Abelardo94, 2019-08-08:
@AndreaBorgia-Abo thank you very much