stanleyhuangyc/Freematics #127: one+ can sniffer

hurukku098, 2021-02-18:
[image]

i uploaded firmware_v5/can_sniffer/can_sniffer.ino
attatched OBD to kia picanto
sniffer is not working and keep saying UART driver already installed

anybody know how to solve?
(elm327 worked in same car)

cworoberts, 2021-04-22:
@hurukku098 I havent use the freematics builder but have experienced the UART driver already installed issue. In my case it was caused by trying to mask more than 8bits of the 11bit header identifier. So obd.setHeaderMask(0xF00) works but obd.setHeaderMask(0xC00) or obd.setHeaderMask(0x800) or really anything other than the F00 causes the UART driver message to be the response. Also, once that error occurs, I can't seem to reset the mask and have to completely remove power from the device for it to reset. If you applied a mask try adjusting to no more than 8bits and also you may need to remove power from the device for at least 1-2 seconds for a reset.

pr414, 2022-12-26:
Hi, was a solution to this found anywhere?

pr414, 2023-03-08:
For the record, I managed to fix this problem by reverting to an older FreematicsPlus library, specifically early 2018 commits. Since my hardware is from late 2017, I had to find by myself that it is no longer supported by the official code.
Creating my own fork of the old commit fixed every issue.