stanleyhuangyc/Freematics #139: Closing car for wakes from standby

wally2511, 2021-04-22:
Closing a car door is causing the device to wake from standby. I've change the MOTION_THRESHOLD to 1.0 and am still experiencing issues.

maxdd, 2021-04-23:
Why are you expecting a MEMS related parameter to have any impact on the device wakeup?
Is it because the door slam is so strong that makes the car roll?
Are you referring to a wake up here

https://github.com/stanleyhuangyc/Freematics/blob/a42b6ab98a296ae81ab4c25a8a5bf4a4b3d3a413/firmware_v5/telelogger/telelogger.ino#L401

https://github.com/stanleyhuangyc/Freematics/blob/a42b6ab98a296ae81ab4c25a8a5bf4a4b3d3a413/firmware_v5/telelogger/telelogger.ino#L732

or is it a different wake up? 
If you can plot the accelerations and those values we might figure out if that's the culprit.

wally2511, 2021-04-23:
Hi @maxdd 

It is the second wake up. The door isn't being slammed that hard although it is a big vehicle and the doors are heavy.

I was thinking it might be worth adding a second condition into the wake up logic? Some like if motion is true and voltage > 14 then start waking up?

maxdd, 2021-04-23:
Hi,
I don't have your use case so i'll try to make some assumptions.
It seems you are afraid of a motion detection even when the car is parked so apparently you are only interested in when the car is moving. I'm assuming you want to use the 14V because the alternator of the car is charging the battery when the engine is running but that would be valid also when you connect a charger to the battery.
Maybe you could try to use engine speed (not so perfect) or engine load but i might be totally wrong since i have checked what the "Motion" detection part is supposed to do.
Long story short, that portion of code is expected to be "modified" for your use case and at that time it might have been the possibility for a car to be "stolen" at engine off (my guess)

wally2511, 2021-04-24:
Thanks for the response @maxdd, I appreciate the fact you are active and willing to help here.

You are correct in what you are saying, the algorithm is seeing the door slam as higher than 1.0g.

**The short story on my use case**
I am looking for a reliable OBD device that can log quality data to an SD card on the vehicle's performance. I am specifically looking to track MAP and MAF data to get insight into a P0106 DTC I've had recently. So, I'm only interested in seeing data about the vehicle when it is running.

To this point I haven't been able to reliably collect any data, so am keen to squash the bugs in the code so I can start getting some quality data.

Below is my solution to the false wakeups from a bump or door closing. It will break the while loop if it experiences two "motions" within 2 seconds or a "motion" and a voltage above 14v.

bool waitMotion(long timeout)
{
#if ENABLE_MEMS
  unsigned long t = millis();
  if (state.check(STATE_MEMS_READY)) {
    do {
      serverProcess(100);
      // calculate relative movement
      float motion = 0;
      float acc[3];
      if (!mems->read(acc)) continue;
      if (accCount == 10) {
        accCount = 0;
        accSum[0] = 0;
        accSum[1] = 0;
        accSum[2] = 0;
      }
      accSum[0] += acc[0];
      accSum[1] += acc[1];
      accSum[2] += acc[2];
      accCount++;
      for (byte i = 0; i < 3; i++) {
        float m = (acc[i] - accBias[i]);
        motion += m * m;
      }
      // check movement
      if (motion >= MOTION_THRESHOLD * MOTION_THRESHOLD) {
        **batteryVoltage = (float)(analogRead(A0) * 12 * 370) / 4095;
        if (millis() - lastMotionTime < 2000 || batteryVoltage > JUMPSTART_VOLTAGE) return true;
        lastMotionTime = millis();**
      }
    } while ((long)(millis() - t) < timeout || timeout == -1);
    return false;
  }
#endif
  serverProcess(timeout);
  return false;
}

maxdd, 2021-04-24:
As a tip use the "code" markers when you are posting code snippets otherwise they are a mess to read.
In any case keep in mind that OBD pids are active requests (that's why you see a lot of disclaimer about safety when driving because they can interfere with the normal ECUs operations) maybe you can try to do a CAN Sniffer (which is passive) and look in github for Mitsubishi Pajero dbc (DataBase Can) which allows you to parse the data

wally2511, 2021-04-24:
Thanks for th tip!

I haven't explored can sniffing yet. Have you played with it on Freematics?

maxdd, 2021-04-25:
I dont own any Freematics device since i wanted to verify the support/functionality.
I'm starting to play with a basic setup and by the time ill have the concept working ill decide whether to add IMU and GPS or not

wally2511, 2021-04-25:
> I dont own any Freematics devices since i wanted to verify the

Are you using an Arduino and a can breakout board? What device would you go
with if you didn't chose a Freematics?

maxdd, 2021-04-28:
I'm currently trying ESP32 + SN65HVD230, but pay attention they are all non-genuine devices so you need to buy a genuine chip (mouse, digi-key) and make your breakout board or resolder a fake one