stanleyhuangyc/Freematics #143: [v5/telelogger] will not send Internal GPS position to server (gd->ts is 0)

TeNNoX, 2021-05-17:
I tried debugging why I didn't receive GPS positon on Traccar server, and I figured out two things:

I had to change the URL to remove the `/push`, as Traccar OsmAnd API is on the base URL.
(Am I using wrong way to connect to Traccar or why is this the case?)

The `sys.gpsGetData(&gd)` would only set `ts` field for non-coprocessor sensor. But in teleclient, it checks `gd->ts` in `transmit`, and otherwise only sends the device ID (probably as a "I'm alive" ping?):
https://github.com/stanleyhuangyc/Freematics/blob/master/firmware_v5/telelogger/teleclient.cpp#L359

And in general I'm uncertain if this is how to use the code correctly with Traccar server, as it required changing the code.

Here's my config:
```cpp
#define SERVER_HOST "demo3.traccar.org"
#define SERVER_PROTOCOL PROTOCOL_HTTPS
#define SERVER_PORT 5055 /* 5170 */

#define SERVER_METHOD PROTOCOL_METHOD_GET
```

soshial, 2021-06-03:
I am not sure if I have the same problem, but I also installed telelogger firmware. [And it constantly sends (0, 0) when it loses signal](https://www.traccar.org/forums/topic/freematics-one-setup-problems/).

As a result I get this route. Is there a way to avoid this?
[image]

bakerkj, 2021-07-30:
Hi, I am seeing this problem as well.

sys.gpsGetData(&gd) is not setting gd->ts, and transmission is conditioned on gd->ts being non zero.

Is there any progress on getting this fixed?

Thanks

stanleyhuangyc, 2021-10-31:
Just fixed this bug.

closed 2021-10-31 by stanleyhuangyc