stanleyhuangyc/Freematics #149: protocol J1939 not suppport by chip ELM327 v 1.5 /STM32F

maicollbellpi, 2021-09-16:
Hola Stanley, soy Maicoll de Bellpi, Colombia. hace unos meses te compramos un dispositivo Freematics ONE+ Model H, que, según la pagina web, soporta el protocolo J1939. revisando tu código y la electrónica del dispositivo, encontré la referencia del chip que controla la etapa de comunicación entre la ESP32 y el CAN BUS del vehículo, este chip se llama ELM327 v 1.5, sin embargo, el datasheet que ofrece el fabricante de este chip, menciona que nunca a desarrollado esta versión del chip. Por lo que decidí a realizar pruebas con tu dispositivo, intercambiando los protocolos que dice soportar, en el config.h y el OBD.h del telelogger.ino Firmware_v.5, testeando desde el 0x1 hasta el 0xC, enviando al chip STM32 el  comando: AT SP h, para configurar el chip al protocolo seleccionado, las pruebas realizadas te las adjunto, para que valide lo que te menciono.

<iframe src="https://casatoro2-my.sharepoint.com/personal/maicoll_bellpi_com/_layouts/15/Doc.aspx?sourcedoc={ff7e4275-cd48-4e4f-b164-5cf84f733fa0}&amp;action=embedview&amp;wdPrint=1&amp;wdEmbedCode=0" width="476px" height="288px" frameborder="0">Esto es un documento de <a target="_blank" href="https://office.com">Microsoft Office</a> incrustado con tecnología de <a target="_blank" href="https://office.com/webapps">Office</a>.</iframe>

debido a que el chip no soporta el protocolo j1939, el ejemplo de J1939_monitor, no entrega resultados.

agradezco tu ayuda

maicollbellpi, 2021-09-16:
[Documento.pdf](https://github.com/stanleyhuangyc/Freematics/files/7179777/Documento.pdf)

stanleyhuangyc, 2021-10-31:
Refer to j1939_monitor example

closed 2021-10-31 by stanleyhuangyc