stanleyhuangyc/Freematics #153: Standby issue, device not waking (i think)

huseyinkogo, 2022-02-15:
Hi Stanley,

It's the trackernsw guy from the forum.

Seems like the device is not waking up from standby.

Having the same issues as the following threads;
https://www.traccar.org/forums/topic/freematics-one-stops-transmitting-after-4-hours-or-so/
https://freematics.com/forum/viewtopic.php?p=48483

Currently one of the vehicles is out for delivery and I can't track it, like the day I inserted the tracker i was able to track them until the 3rd or 4th trip of the day, and then it is no update.

 Today, I will track myself again, by unplugging and replugging the tracker each trip, to see if its the standby issue

huseyinkogo, 2022-02-16:
UPDATE:
The tracker is working if the device is unplugged & replugged when the vehicle is on after the vehicle was off. So the problem is the wake up after standby

huseyinkogo, 2022-02-22:
@stanleyhuangyc 

Could I please could i get an update?

There should be a way, I don't want to unplug & replug the GPS tracker in order it to work.

Happy to sponsor this issue

brentk7, 2022-08-24:
I have the same issue with one of my 2 trackers. One works fine all the time - the second stops sending location updates (although it does still regularly connect to traccar server) until unplugged and plugged back in.
Both are configured the same (apart from the fact one has internal GNSS and the other external)

ankostis, 2022-09-06:
@huseyinkogo can you try [this deadlock fix (6d30abf4)](https://github.com/ankostis/Freematics/commit/6d30abf45c0f8eb59590cd6b1a16c35ff7cbc6fa) to see if it fixes your wake-up issue?

ankostis, 2025-05-17:
I'm reporting here what is the root cause of this stochastic deadlock in `telemetry()`, apparent in older versions of the firmware but not anymore, and why it could in principle manifest in other places.

### Root cause
The high-priority(2) `telemetry()` task would deadlock in state 3C (not `WORKING`, but not yet `STANDBY`) right before falling to sleep, because its outer-loop would roll in quick succession without any `delay()` to allow for the lower-priority(1) `loop()` to task-switch and let `process()` to return, setting thus the `STANDBY` flag to 1.

### Why it has been workarounded?
I guess in later versions of firmware (after the big _submerged_ commit 0870d18 became public, circa summer 2021), a _check for WiFi_ intervened, which gave the opportunity to FreeRTOS to switch to lower priority tasks:
https://github.com/stanleyhuangyc/Freematics/blob/1ce521bfa93496033cee0ea58bc712473c94a4bb/firmware_v5/telelogger/telelogger.ino#L905-L919

### Proper fix(?)
The proper solution though would be to equalize the priority of both tasks to 1 (which is Arduino's default priority of `loop()`):
https://github.com/stanleyhuangyc/Freematics/blob/1ce521bfa93496033cee0ea58bc712473c94a4bb/firmware_v5/telelogger/telelogger.ino#L1473
This would eliminate the possibility for the bug to re-emerge, if the state-machine code is modified in the future.

Is there a counter-reason not to equalize priorities?