stanleyhuangyc/Freematics #160: Info: how I managed to read battery PIDs from my EV

bobobo1618, 2022-05-23:
I have a Kia EV6 (same platform as the Hyundai Ioniq 5) and wanted to get the Freematics One+ Model B reading data from it. I've managed that and thought I'd share. I'd share on the forum but I'm waiting for a moderator to approve my posts for around a week now.

I prefer working with the ESP-IDF so I don't have code to contribute to the Freematics repo but here's what I did to get useful data out of my EV:

1. Figure out which ECU to talk to and which PIDs contain information I'm interested in. Folks have done a lot of work for the Hyundai Ioniq 5 so I just needed to look at [this CSV](https://github.com/Esprit1st/Hyundai-Ioniq-5-Torque-Pro-PIDs/blob/main/TorqueIONIQ5AWD77kWh.csv). For battery information, I need to send PID `220101` to ECU `7E4`.
2. Figure out which protocol the car uses. [These guys](https://www.autopi.io/electric-vehicle-types/hyundai-ioniq-5-integrate-with-autopi/) say it uses ISO 15765-4, an 11-bit header and 500000 baud rate.
3. Use all this to talk to the OBD chip (I used [the ELM327 manual](https://www.elmelectronics.com/wp-content/uploads/2020/05/ELM327DSL.pdf) for a lot of guidance):

```
// Reset the chip to start from a clean slate
ATZ
// Set the protocol to "ISO 15765-4 CAN (11 bit ID, 500 kbaud)" to match the car. The number "6" comes from page 27 of the ELM327 manual.
ATSP6
// Set up the CAN filter and mask to only allow responses from OBD-II ECUs. This follows the example on page 49 of the ELM327 manual.
ATCF7E8
ATCM7F8
// Tell the OBD chip to talk to the BMS, which has address 7E4 as discussed above
ATSH7E4
// We're now ready to start getting data
220101
// Data comes now...
```

Looking at the Freematics code, if you want to use the Freematics code to do the same thing, you need to:
- Hard-code a protocol to [trigger the `ATSP`](https://github.com/stanleyhuangyc/Freematics/blob/a72f876f993ad76021a8272b23dcac867d158617/libraries/FreematicsPlus/FreematicsOBD.cpp#L369-L374). For example in the `telelogger` example, you'd need to put `PROTO_CAN_11B_500K` [here](https://github.com/stanleyhuangyc/Freematics/blob/a72f876f993ad76021a8272b23dcac867d158617/firmware_v5/telelogger/telelogger.ino#L498)
- Call the [setHeaderFilter and setHeaderMask functions](https://github.com/stanleyhuangyc/Freematics/blob/a72f876f993ad76021a8272b23dcac867d158617/libraries/FreematicsPlus/FreematicsOBD.cpp#L478-L494) (I suppose I'd add these calls around the [OBD init code](https://github.com/stanleyhuangyc/Freematics/blob/a72f876f993ad76021a8272b23dcac867d158617/firmware_v5/telelogger/telelogger.ino#L499))
- Implement a fork of [readPid](https://github.com/stanleyhuangyc/Freematics/blob/a72f876f993ad76021a8272b23dcac867d158617/libraries/FreematicsPlus/FreematicsOBD.cpp#L75-L106) that supports multi-byte PIDs and responses

ankostis, 2023-06-22:
> * Implement a fork of [readPid](https://github.com/stanleyhuangyc/Freematics/blob/a72f876f993ad76021a8272b23dcac867d158617/libraries/FreematicsPlus/FreematicsOBD.cpp#L75-L106) that supports multi-byte PIDs and responses

Can you give an example response, in bytes?

DutchessNicole, 2025-08-17:
Could you please post some examples of how to exactly do these steps? The listed steps are way too vague for me to make anything useful from them.