stanleyhuangyc/Freematics #165: Teleogger goes into sleep/stanby prematurely

TandemPhoenix, 2022-07-23:
When at idle for slightly longer than normal times device will enter sleep/standby.

Would like to adjust the time before the device changes states.

Alternatively, and even more preferred, I would like an option to have the device enter sleep/standby only after no longer receiving data from OBD.

ankostis, 2022-07-23:
To prolong the standby tumor increase the last value of the `STATIONARY_TIME_TABLE` 3-tuple:
https://github.com/stanleyhuangyc/Freematics/blob/28c82e18fa4e55ec9cb5fc0d092a189046310de6/firmware_v5/telelogger/config.h#L99

Unfortunately, OBD never stops sending data, even if the engine is switched off - that is why different tricks are employed (MEMS detecting motion, supply battery voltage spike when the engine is switched on).

ankostis, 2022-07-27:
Granted that you're satisfied with response, i guess you may close this issue.

ankostis, 2022-12-19:
@TandemPhoenix will you close this issue?

closed 2022-12-21 by TandemPhoenix