stanleyhuangyc/Freematics #173: firmware_v5/telelogger Problem bulding

roysbike, 2022-12-06:
PlatformIO Core, version 6.1.5

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
;platform = https://github.com/platformio/platform-espressif32.git
platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.5
;build_flags = -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue
board=esp-wrover-kit
board_build.f_cpu = 160000000L
framework = arduino
monitor_speed = 115200
board_build.flash_mode = dio

```
;[env:esp32c3]
;platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
;platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.5
;  toolchain-riscv-esp
;  platformio/tool-esptoolpy
;framework = arduino
;board = esp32-c3-devkitm-1
;board_build.mcu = esp32c3
;monitor_speed=115200

[platformio]
src_dir=.
```

[env]
lib_extra_dirs=../../libraries

```
Compiling .pio/build/esp32dev/src/telelogger.ino.cpp.o
Retrieving maximum program size .pio/build/esp32dev/firmware.elf
Checking size .pio/build/esp32dev/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.6% (used 18440 bytes from 327680 bytes)
Flash: [===       ]  29.1% (used 380921 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, ftdi, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/cu.usbserial-01E73F49
Uploading .pio/build/esp32dev/firmware.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: '/Users/roys/.platformio/packages/framework-arduinoespressif32@src-f2ea83e2545300b10a69ff44ef9dc6cd/tools/sdk/esp32/bin/bootloader__40m.bin'
*** [upload] Error 2
```

wladkolc, 2022-12-19:
Hi, it is reported as bug in arduino esp32 framework, just change 2.0.5 version to 2.0.3 and delete previous framework, build and upload again.

platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.3

voklav, 2023-02-13:
Thanks for the guidance!
I wrote a [new post](https://github.com/platformio/platform-espressif32/issues/1028#issue-1581794812) before reading this :/ but the problem seems to persist.