stanleyhuangyc/Freematics #175: schematics diagram / known quirks for the One+

skhopkins, 2022-12-19:
I'm experimenting with Micropython on the One+ as an alternative to the current codebase however trying to work backwards through the existing code is somewhat of a challenge, especially as some of the documentation/examples don't appear to align with testing. For example, following https://freematics.com/pages/products/freematics-one-plus/guide/ shows the One+ using GPIO's 26/34 for the external GPS (in the picture and the text), yet the One+ header file for the v5 firmware shows it to be 32/33 (which works).

Is there a schematics diagram available for the One+ that details what connects to each GPIO pin? Thanks!

ankostis, 2022-12-19:
Doesn't [this schematics apply](https://github.com/ankostis/Freematics/blob/main/firmware_v5/schematics_oneplus_hd_20200604.pdf) to your device?  (downloaded from the site)

I can contribute with that task. Do you plan to share your repo with GNU license?

skhopkins, 2022-12-19:
Hi @ankostis looking at that diagram I don't believe so as the GPIO 32/33 for the GPS don't map to those pins.
Any code I create around this will be public, will need to check the license of linked code such as the micropython async GPS etc.

ankostis, 2022-12-19:
.  yet the One+ header file for the v5 firmware shows it to be 32/33 (which works).

In the diagram i sent you GPIO-pins 26,34 are shown indeed connected to the external GNSS - can you paste here the link of the schematics you're talking about where it shows 32/33?
I'm a somewhat confused of what you describe.

skhopkins, 2022-12-19:
In the header file for the plus model (https://github.com/stanleyhuangyc/Freematics/blob/master/libraries/FreematicsPlus/FreematicsPlus.h#L66) it has the following:

#define PIN_GPS_POWER2 15
#define PIN_GPS_UART_RXD2 32
#define PIN_GPS_UART_TXD2 33

Those three pins work on my model for the external GPS. Pins 26/34 are also referenced however I believe there is a model check in the code which determines which pins to use.

calvarezm1, 2023-02-16:
It depends on the model you're using, accordingly to freematics documents the ones that incorporates a GNSS ceramic antenna embedded use the GPIOs 15, 32 and 33 to communicate with it and leave the GPIOs 26 and 34 available via de molex connector. 

The schematic ankostis shared is correct for any model, the only difference is on the software where depending on the model it configures the GPS uart using PIN_GPS_UART_RX/TX or GPS_UART_RX2/TX2.