stanleyhuangyc/Freematics #183: Firmware V5 Telelogger posts only zeroes to traccar

arduino-man, 2023-06-05:
I recently installed the latest Telelogger V5 firmware into a Freematics One+ H but looking at the received lat: and lon: on Traccar, things look like this:

Traccar Log:
```
2023-06-04 22:54:22  INFO: [U65a04d6d: freematics < 74.198.149.120] 5543464c4641594d23303a33353832322c32343a3336352c32303a303b303b302c31303a32323534323230332c38323a34332a4232
2023-06-04 22:54:22  INFO: [U65a04d6d] id: UCFLFXXX, time: 2023-06-04 22:54:22, lat: 0.00000, lon: 0.00000, course: 0.0
2023-06-04 22:54:28  INFO: [U65a04d6d: freematics < 74.198.149.120] 5543464c4641594d23303a34313539342c32343a3336372c32303a303b303b302c31303a32323534323738392c38323a34332a4341
2023-06-04 22:54:28  INFO: [U65a04d6d] id: UCFLFXXX, time: 2023-06-04 22:54:27, lat: 0.00000, lon: 0.00000, course: 0.0
2023-06-04 22:54:31  INFO: [U65a04d6d: freematics < 74.198.149.120] 5543464c4641594d23303a34343438312c32343a3336372c32303a303b303b302c31303a32323534333037322c38323a34342a4242
2023-06-04 22:54:31  INFO: [U65a04d6d] id: UCFLFXXX, time: 2023-06-04 22:54:30, lat: 0.00000, lon: 0.00000, course: 0.0
2023-06-04 22:54:33  INFO: [U65a04d6d: freematics < 74.198.149.120] 5543464c4641594d23303a34373336382c32343a3336372c32303a303b303b302c31303a32323534333336352c38323a34342a4337
2023-06-04 22:54:33  INFO: [U65a04d6d] id: UCFLFXXX, time: 2023-06-04 22:54:33, lat: 0.00000, lon: 0.00000, course: 0.0
```

Freematics Debug Log (VScode):
```
RSSI:-69dBm
[GPS] 59.125103 -119.158127 0km/h Course:0 2023-06-04T22:56:46.600Z
[BUF] 5 samples | 46 bytes | 1/1024
UCFLFXXX#0:180488,81:-69,24:367,20:0;0;0,10:22564667,82:45*6C
[CELL] 61 bytes being sent
[NET] 02:44.3 | Packet #43 | Out: 2 KB | In: 47 bytes
[GPS] 59.125103 -119.158127 0km/h Course:0 2023-06-04T22:56:51.700Z
[BUF] 4 samples | 38 bytes | 1/1024
UCFLFXXX#0:185490,24:359,20:0;0;0,10:22565172,82:45*F8
[CELL] 54 bytes being sent
[NET] 02:49.3 | Packet #44 | Out: 2 KB | In: 47 bytes
RSSI:-75dBm
[GPS] 59.125107 -119.158127 0km/h Course:0 2023-06-04T22:56:56.700Z
[BUF] 5 samples | 46 bytes | 1/1024
```

I'm not sure how to proceed.

arduino-man, 2023-06-05:
I've narrowed it down to this line in telelogger.ino:
```
 if (gd->lat && gd->lng && gd->sat > 3)
```
Changing the condition to be:
```
if (gd->lat && gd->lng)
```
Fixed the issue for me.

closed 2023-06-05 by arduino-man

ankostis, 2023-06-06:
What about the quality of Lat/Lon signalt?

- Do they make rense on a map?
- Are you using external GPS?

arduino-man, 2023-06-06:
I'm using the internal GPS antenna and yes, Lat/Lon both look good.