stanleyhuangyc/Freematics #184: EV support for Freematic libraries

PonomarovP, 2023-06-06:
Standard OBD-2 protocol is not supported in electric vehicles. Other diagnostic protocols over OBD-socket are used instead, applying longer parameter IDs. Is there a plan to support these protocols in further Freematic software versions?

JeroenVanOort, 2023-06-21:
I doubt it's even specific to EV's. It might simply be recent cars in general: they don't support the standard OBD-II PID's, but work using [Unified Diagnostic Services](https://en.wikipedia.org/wiki/Unified_Diagnostic_Services) instead.

Based on the information in [this issue](https://github.com/stanleyhuangyc/Freematics/issues/160) I've been trying some things and these are some of the problems in need of solving in order to support UDS.

1. Using other services than `01`.
    Currently the service / mode used is [hardcoded](https://github.com/stanleyhuangyc/Freematics/blob/e831f9712172504769d5dfb654c3114a85bd591b/libraries/FreematicsPlus/FreematicsOBD.h#L63).
2. Better response parsing
    I believe that at this time, only the first line returned by the co-processor (acting as an ELM327) is parsed, while there may be several, as UDS uses [ISO-TP](https://en.wikipedia.org/wiki/ISO_15765-2) in order to handle messages that are larger than the space available in a CAN frame.
3. Support getting multiple data points from one PID
    In UDS, it seems common to have one PID with many data fields within it. For example `220101` in Hyundai/KIA vehicles:
    [image]
    [source](https://docs.google.com/spreadsheets/d/1eT2R8hmsD1hC__9LtnkZ3eDjLcdib9JR-3Myc97jy8M/edit#gid=246048743)

I do think the lack of UDS support will increasingly be a deal breaker for Freematics products.

stanleyhuangyc, 2023-08-05:
We are considering adding ISO-TP.

ftastisch, 2025-07-07:
Did this actually happen?