stanleyhuangyc/Freematics #186: Corrupt GPS data after upgrading to "Improved GNSS module initialization" commit

gyrofx, 2023-09-17:
After using the latest telelogger build (https://github.com/stanleyhuangyc/Freematics/commit/1505b85834847aa0922b6f53effabc757b737b1f) GPS data coming from the device seems corrput.

[image]

The `date` field is always `000000`, which causes parsing to fail and no GPS data to be sent or recorded.

Without the changes from this commit, everything works as expected.

[image]

Device: Freematics One+ Model B (pretty new), internal GPS, WIFI, built with platform.io on Windows

canegru, 2023-09-19:
Confirmed as well, I reverted the commit and it works as expected.

stanleyhuangyc, 2023-10-06:
This commit seems have affect older batch of devices. Will fix this ASAP.

stanleyhuangyc, 2023-10-06:
Avoided M10 GNSS commands from being sent as a workaround.

dd310, 2023-10-06:
> This commit seems have affect older batch of devices. Could you guys confirm the date of purchase of your device?

09/09/23

canegru, 2023-10-06:
> This commit seems have affect older batch of devices. Could you guys confirm the date of purchase of your device?

09/12/23 here

gyrofx, 2023-10-06:
> This commit seems have affect older batch of devices. Could you guys confirm the date of purchase of your device?

07/28/23

stanleyhuangyc, 2023-10-08:
Fixed now. Please try latest commit.
The issue is caused by date value is no longer reported after new GNSS commands sent but date is still checked in the code.

stanleyhuangyc, 2023-10-08:
The issue is fixed now.

The cause:
Recently added m10 configuration commands disable GNSS' RMC command so no date value is reported but date value is checked and non-zero required in the library code.

dd310, 2023-10-11:
Just tried the latest commit and the issue seems indeed solved. Thanks!

closed 2023-10-15 by stanleyhuangyc