stanleyhuangyc/Freematics #189: Freematics ONE+ not reporting additional PIDS

wladkolc, 2023-10-26:
I have tried to add all PIDS, only 5 at one time, trying different priorities, and device not reporting it to traccar. Few months ago my car reported pid oil temp properly, but after some time (dont know exactly) it stopped working and it reporting only predefined "default" pid that was in script. I have tried everything to add into telelogger.ino and it not works.

I am adding it like this:

PID_POLLING_INFO obdData[]= {
  {PID_SPEED, 1},
  {PID_RPM, 1},
  {PID_THROTTLE, 1},
  {PID_ENGINE_OIL_TEMP, 1},
  {PID_FUEL_PRESSURE, 2},
  {PID_TIMING_ADVANCE, 2},
  {PID_COOLANT_TEMP, 3},
  {PID_INTAKE_TEMP, 3},
};

and result is missing PID_ENGINE_LOAD

| 621173 | freematics |        4 | 2023-10-26 09:44:23 | 2023-10-26 09:44:22 | 2023-10-26 09:44:22 | 0x01         | 49.203697 | 18.734217 |    368.2 |         0 |      0 | NULL    | {"obdSpeed":0.0,"rpm":0,"throttle":15,"battery":13.19,"acceleration":"-0.01;0.08;0","sat":11,"hdop":94,"deviceTemp":5.5,"distance":0.15,"totalDistance":2878221.98,"motion":false} 

I am desperated, how much problems i have with this device.

Anybody know how it works?

stanleyhuangyc, 2023-10-31:
PID_ENGINE_LOAD is not in the array.

closed 2023-10-31 by stanleyhuangyc

wladkolc, 2023-10-31:
It was only as example, nothing except predefined PIDs not working. I have tried it read with ELM and it reads another PIDS.