stanleyhuangyc/Freematics #196: Freematics One+ B - Unable to resolve GPS data using newer library

ZacharyDuBois, 2024-01-15:
Using the newest Telelogger example code, I can no longer use the GNSS or MEMS on my older One+ B. Looking at the code, it looks like a lot of the code pertaining to the older model was deleted (ublox M8 / MPU-9250). I was able to get the MPU-9250 to work again by uncommenting some code in `telelogger.ino` but I am unable to get the M8 to register anymore as it looks like it was completely rebuilt.

```
CPU:160MHz FLASH:16MB
IRAM:343KB
DEVICE ID:[redacted]
TYPE:14
MEMS:MPU-9250
ACC BIAS:0.00/-0.12/0.88
GNSS:NO
OBD:OK
SD:14752 MB total, 319 MB used
File: /DATA/13.CSV
VIN:[redacted]
[CELL] Activating...
[BUF] 8 samples | 70 bytes | 1/32
CELL:SIM7600A-H
IMEI:[redacted]
[CELL] Searching...
Operator:T-Mobile
[CELL] IP:[redacted]
LOGIN(am[redacted]:8081)...
[BUF] 32 samples | 280 bytes | 4/32
```

Note, before my MEMS modification, the MEMS line would return "NO".

Please support the older model One+ B. I really would love to not have these become ewaste.

stanleyhuangyc, 2024-03-18:
If your ONE+ has an older GNSS module, comment out line 550 and uncomment line 552 in FreematicsPlus.cpp.

stanleyhuangyc, 2024-03-18:
What is the device ID you have got?

sn3ak, 2024-07-26:
Thanks for sending me down the right path. For anyone else who has the same problem, you need to uncomment the code at [this link](https://github.com/stanleyhuangyc/Freematics/blob/06fb90ed6c77ea8994bfdf39a12d8682d58ecbb1/firmware_v5/telelogger/telelogger.ino#L1429).

I managed to get my GPS working. I thought I should share my findings.

First, Comment this [line](https://github.com/stanleyhuangyc/Freematics/blob/06fb90ed6c77ea8994bfdf39a12d8682d58ecbb1/libraries/FreematicsPlus/FreematicsPlus.cpp#L551) and uncomment this [line](https://github.com/stanleyhuangyc/Freematics/blob/06fb90ed6c77ea8994bfdf39a12d8682d58ecbb1/libraries/FreematicsPlus/FreematicsPlus.cpp#L553). Next, modify this [line](https://github.com/stanleyhuangyc/Freematics/blob/06fb90ed6c77ea8994bfdf39a12d8682d58ecbb1/libraries/FreematicsPlus/FreematicsPlus.cpp#L601) and replace "$GNGGA" with "$G". I know for sure the second change is required. I am not sure about the first, but since it says it's for the m8 I figured I would run with it.

Note: If you are using the telelogger package this file is located under main\libraries\FreematicsPlus. 
I hope this helps!