stanleyhuangyc/Freematics #197: pidmap always set to 0xff

i-like-rockets, 2024-04-01:
https://github.com/stanleyhuangyc/Freematics/blob/8612784dbb481933b9e356d6c8cbf3e483a27360/libraries/FreematicsPlus/FreematicsOBD.cpp#L412

Since the pidmap reference in #399 `memset(pidmap, 0xff, sizeof(pidmap));` sets all values to `0xff`, there seems to be no instance where this `|=` would ever equate to anything except `0xff`.

If this interpretation is correct, either this sign should changed to `=` (default to keep PID in map) or the `memset` command should be removed (default to ignore PID).

ankostis, 2024-04-02:
Unfortunately there is no useful [commit message](https://github.com/stanleyhuangyc/Freematics/commit/b5f6f1f8e49db37dfe20df7f300914049bcbca15) to explain why `|=` was necessitated, 3yago.

stanleyhuangyc, 2024-04-03:
Filling pidmap to 0xff is to assume all PIDs are available until actual PID mapping data is loaded.

stanleyhuangyc, 2024-04-03:
I checked this seems to be a typo as the same code is right in the library for OBD-II UART adapter. It really should be = instead of |=.

stanleyhuangyc, 2024-04-03:
It seems to be truly an unnecessary change 3 years ago while the same code is right in the library for OBD-II UART adapter. It really should be = instead of |=.

stanleyhuangyc, 2024-04-03:
This issue is fixed now and thanks for the finding.

closed 2024-04-03 by stanleyhuangyc