stanleyhuangyc/Freematics #200: Option to send location with pings.

Cyberes, 2024-07-03:
It would be nice to have an option to send location data with pings. I know it kind of defeats the purpose of lightweight pinging, but I would like to ensure that nothing goes wrong between the tracker and the server.

archer411, 2024-07-03:
i would like this option as well. my cell plan has quite a lot of data so i have no reason to conserve it with locationless pings

Cyberes, 2024-07-14:
I tried to implement something based off of this commit: https://github.com/soshial/Freematics/commit/f5c052c790073799485c560d8738793e54b37498

But the current codebase has changed since then and I don't have enough experience with C.

stanleyhuangyc, 2024-07-21:
This is possible to implement but will more likely to drain car battery.

Cyberes, 2024-08-09:
That's fine, just keep it off by default.

Cyberes, 2024-08-22:
@stanleyhuangyc what do you think about having a configurable interval to send location data when the device/car is parked and in sleep mode. I'm thinking something like once an hour.