stanleyhuangyc/Freematics #202: GNSS settings in new version

malagsch, 2024-08-06:
Hi

because there is not really a documentation i was able to find im asking here in the community.
With the old version i had "GGNS_EXTERNAL"
now this option is removed in the config.h
new version has
#define GNSS_NONE 0
#define GNSS_STANDALONE 1
#define GNSS_CELLULAR 2

i selected standalone but no gps signal
i have model A+ with external GPS antenna

sn3ak, 2024-08-10:
Support for 2020 units has been commented out or removed. 
Check my comment [here](https://github.com/stanleyhuangyc/Freematics/issues/196#issuecomment-2251860047) and see if it helps you. I've had to do some more tinkering since and am still trying to dial things in, but I have it working fairly reliably.

malagsch, 2024-09-24:
thanks @sn3ak 
i know you replied some weeks ago but yesterday i had the chance to test it and it worked.

closed 2024-09-24 by malagsch